Mat931/esp32-doorbell-bus-interface #1: [Feature Request] plain board design without esp32

m-idler, 2024-02-02:
I'm really keen on the project and would love to try it out as soon as possible. One thing that I'm struggling with, is the possibility to either use it with an existing esp32/esphome or somehow get a wired ethernet connection available (i would prefer to install it close to the welcome module in the distribution cabinet). As I understood so far, connecting an ethernet chip via pins/wires wouldn't be a very reliable solution, so i guess a generic board without an hardwired esp, just with a pin interface for external boards would be really helpful. Unfortunately I haven't any experience in circuit board design / modification yet and expect it to be a pretty error-prone way for a newbie. Maybe someone could help here to make this really cool hardware and the implementation in esphome available to even more people.
Thank you

Mat931, 2024-02-03:
I designed the board so that it can work and receive power from the bus without any external cables. If you want to connect things like ethernet or external power I don't know what effects that will have on the ABB-Welcome system (maybe reduced video and audio quality or worse?) so you need to consider isolating the bus interface from the rest of your circuit. In my experience the wifi connection of the ESP32 is reliable enough for this project, that's why I picked it over an ESP8266. Not sure what you have in mind for your "generic board". Should it supply power from the bus or are you planning to use an external supply? I guess you could make a minimal board with just the transmit and receive circuit, protection and two optoisolators.

m-idler, 2024-02-03:
Yes i think that would be a good idea to have a minimal board, that just contains the transmit and receive circuit, protection , the optoisolators and pin connectors to connect the rx/tx pins (and maybe other needed pins) to a separate esp32 (e.g. one of the olimex boards with included ethernet). 
Unfortunately i don't have a licence of altium designer to (at least) try stripping down the current board to a minimal one. I tried importing it to KiCad - PCB Editor but unfortunately just get an error "Wrong file format". :(

Mat931, 2024-02-03:
I used [EasyEDA](https://easyeda.com/editor) to design the circuit. The other file formats are just exported from there. I think you can even use it without an account if you upload the EasyEDA JSON files. I just noticed that I forgot to export and upload the schematic file, it needs to be exported separately. Now all the required files are in the repo.

LorbusChris, 2024-05-03:
Awesome project, thanks @Mat931!

You may want to consider the [Feather / FeatherWing form factor]( https://learn.adafruit.com/adafruit-feather/feather-specification ) for an MCU-agnostic board.

Feathers are the the MCU base boards. There are many Feather spec-compatible boards with different MCUs available from different manufacturers, e.g. Adafruit, Sparkfun (Thing Plus), ThingPulse (ePulse), and more.

FeatherWings are the add-on boards that stack on top for io extension (even multiple).
A quick search turned up that there are already a [CAN Bus FeatherWing](https://learn.adafruit.com/adafruit-can-bus-featherwing/overview), [M-Bus FeatherWing](https://www.hwhardsoft.de/deutsch/projekte/m-bus-featherwing/) and even a [Power-over-Ethernet FeatherWing](https://silicognition.com/Products/poe-featherwing/) as an example of one that powers the Feather from the bus.

nils1981, 2024-10-30:
Very nice project - thank you for sharing! As the Minimum Order quantity is 2 or 5 - someone in Germany wants to sell a board?

Mat931, 2024-11-07:
@Haych One other thing my PCB does is power the ESP32 from the doorbell bus. If you're using an external power supply it might have unintended side effects like worse video and audio quality, I haven't really tested that.

Mat931, 2024-11-07:
> I don’t suppose video and audio integration is on your roadmap?
Not for this project. I meant the video and audio quality of your or other indoor stations. The signal is sent in analog form over the same two wires. Electrical noise from a power supply might make it worse.

Chillidaddy, 2024-11-12:
> Very nice project - thank you for sharing! As the Minimum Order quantity is 2 or 5 - someone in Germany wants to sell a board?

Germany here too. Maybe a group buy?

RoMaTiX99, 2024-11-14:
Hey @Chillidaddy , @nils1981 . Just saw your comments - I'm from Germany too and highly-interested. 
So already 3 of us. Group buy sounds great. 

Costs would be about 75€ in total. /3 makes roughly 25€ p.P. Sound's great, isn't it? 
Should we do it?

nils1981, 2024-11-14:
Hi @RoMaTiX99, Hi @Chillidaddy 
yes - I am very much interested. 
I will contact you seperately with the email from your profile.

nils1981, 2024-11-23:
Hi @FloMeyer,
There is 1 last board left. Expecting delivery next week.

FloMeyer, 2024-11-23:
> Hi @FloMeyer, There is 1 last board left. Expecting delivery next week.

Nice, i‘ll take ist. Can you sent me an email? Do you see it in my Profile? (Should be public)

LorbusChris, 2024-11-23:
I have one spare device that I'd sell at cost. Feel free to send an email (see my git commits).

BreBarMevco, 2025-05-17:
Hey all great project and thanks @Mat931 

My boards have just arrived in Australia, I have 4 spares also happy to sell at cost + shipping.

Anyone who needs one yell out!

BreBar07, 2025-05-17:
Hey all, first thanks @Mat931 amazing project.

Just received my 5 complete boards, I only need one.

Anyone who needs one yell out happy to sell at cost + shipping, I’m based in Australia.

Thanks!