Mat931/esp32-doorbell-bus-interface #2: [Feature Request] Enable wiki on the repo to better document the protocol

jmoransalama, 2024-02-07:
First of all, I want to thank the author for taking the time and effort to do this fantastic project. I've been using it for a few weeks and it works wonderfully.

I've tried to find documentation on how the communication protocol works, but I have not been able to find much information.

I think it would be useful if the repository wiki was activated so that people can add everything they discover about the protocol.

For now the little information I have found is the following:

Message Types:
```
DoorbellOutdoor = 0x01,
EndCall = 0x02,
AcceptCall = 0x03,
ManualVideo = 0x04,
Light = 0x09,
Call = 0x0a,
Door = 0x0d,
DoorbellIndoor = 0x11,
Actor = 0x16,
Video = 0x52,
```
 
Device Types:
```
IndoorStation = 0x1,
OutdoorStation = 0x2,
Gateway = 0x3,
Relay = 0x4,
Actor = 0x8,
Video = 0x9,
```

My system also sends messages with the following message type codes: 0x0B, 0xA0, 0x20, 0x60, 0xA2 but I don't know what they mean.

Mat931, 2024-02-07:
Great idea to start a wiki! I'm not sure about the data format either, these are just guesses what the messages could mean. Some more reverse engineering is definitely needed.

Mat931, 2024-02-07:
I added some of the message types I found [here](https://github.com/Mat931/esp32-doorbell-bus-interface/wiki/Message-Types). If you have some new info or a guess what messages could mean please tell me!

mallorca2288, 2024-02-07:
I've found that changing the password to open the door on the indoor unit will send a 0x0F message with the password in the Data. I don't get any reply to this message.

For example, setting the password to 123456 will send the following message:
`55.FF.46.0F.FF.02.FF.00.01.08.C0.02.01.12.34.56.FF.03 (18) [000108 > FF02FF] Type: 0F, Data: 02.01.12.34.56.FF (6)`

Setting the password to 555555 will send the following message:
`55.FF.46.0F.FF.02.FF.00.01.08.35.02.01.55.55.55.FF.A8 (18) [000108 > FF02FF] Type: 0F, Data: 02.01.55.55.55.FF (6)`

Disabling the password:
`55.FF.46.0F.FF.02.FF.00.01.08.E8.02.00.55.55.55.FF.1C (18) [000108 > FF02FF] Type: 0F, Data: 02.00.55.55.55.FF (6)`

So the second part of the data tells the "controller" if the password is enabled = 01 or disabled = 00 and the third, fourth and fifth parts are the password.

mallorca2288, 2024-02-07:
0x1A message is sent together with a 0x0A message when calling the guard-unit from the indoor unit.

In my building there isn't a guard-unit, I suppose that's why I don't get any response.

dgomes, 2024-02-11:
Some more information to help sort the protocol out:

I have a single home systems composed of controller, outdoor unit and indoor unit.

The following debug is a session in which the outdoor unit started a call, the indoor unit answered and later pressed the unlock button.

```
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.42.0A.00.01.01.00.02.01.2B.01.15.D9 (14) [000201 > 000101] Type: 0A, Data: 01.15
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.42.0B.00.01.01.00.02.01.2B.81.15.72 (14) [000201 > 000101] Type: 0B, Data: 81.15
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.42.8B.00.02.01.00.01.01.2B.01.15.BA (14) [000101 > 000201] Type: 8B, Data: 01.15
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.42.8A.00.02.01.00.01.01.2B.8B.15.13 (14) [000101 > 000201] Type: 8A, Data: 8B.15
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.45.01.00.01.01.00.02.01.9D.28.15.01.1E.11.ED (17) [000201 > 000101] Type: 01, Data: 28.15.01.1E.11 (5)
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.44.81.00.02.01.00.01.01.9D.29.01.1E.11.9D (16) [000101 > 000201] Type: 81, Data: 29.01.1E.11
[17:00:37][I][remote.abbwelcome:110]: Received: 55.FF.4A.23.00.20.FF.00.02.01.F7.02.00.00.00.00.00.00.00.00.00.7F (22) [000201 > 0020FF] Type: 23, Data: 02.00.00.00.00.00.00.00.00.00 (10)
[17:00:40][I][remote.abbwelcome:110]: Received: 55.FF.45.01.00.02.01.00.01.01.D4.00.15.03.78.11.6F (17) [000101 > 000201] Type: 01, Data: 00.15.03.78.11 (5)
[17:00:40][I][remote.abbwelcome:110]: Received: 55.FF.44.81.00.01.01.00.02.01.D4.29.03.78.11.15 (16) [000201 > 000101] Type: 81, Data: 29.03.78.11
[17:00:45][I][remote.abbwelcome:110]: Received: 55.FF.41.0E.00.04.01.00.01.01.D0.00.79 (13) [000101 > 000401] Type: 0E, Data: 00
[17:00:45][I][remote.abbwelcome:110]: Received: 55.FF.41.8E.00.01.01.00.04.01.D0.00.95 (13) [000401 > 000101] Type: 8E, Data: 00
[17:00:45][I][remote.abbwelcome:110]: Received: 55.FF.44.35.00.13.01.00.02.01.2F.00.02.01.01.0D (16) [000201 > 001301] Type: 35, Data: 00.02.01.01
[17:00:46][I][remote.abbwelcome:110]: Received: 55.FF.C4.35.00.13.01.00.02.01.2F.00.02.01.01.EE (16) [000201 » 001301] Type: 35, Data: 00.02.01.01
[17:00:46][I][remote.abbwelcome:110]: Received: 55.FF.C4.35.00.13.01.00.02.01.2F.00.02.01.01.EE (16) [000201 » 001301] Type: 35, Data: 00.02.01.01
[17:00:50][I][remote.abbwelcome:110]: Received: 55.FF.41.02.00.02.01.00.01.01.0F.00.12 (13) [000101 > 000201] Type: 02, Data: 00
[17:00:50][I][remote.abbwelcome:110]: Received: 55.FF.40.82.00.01.01.00.02.01.0F.D9 (12) [000201 > 000101] Type: 82
```

0x0201 is the outdoor unit
0x0101 is the indoor unit

don't know what is 0x0401 nor 0x1301

mallorca2288, 2024-02-11:
> don't know what is 0x0401 nor 0x1301

I think 0x000401 might be the address of the actuator unit that unlocks your door.

For example this is what unlocks the doors in my apartment:
[image]

dgomes, 2024-02-11:
pretty sure I don't have that ... the door is connected to the outside unit directly

mallorca2288, 2024-02-11:
I just checked my installation and some doors are connected directly to the outdoor unit as you indicate.

In units where the door is directly connected without a separate actuator, the same unit has different addresses for different functions.

For example, my south outdoor unit has the following addresses:
3E0204 to make/receive calls (Code 0x01 and 0x81)
3E0404 to unlock the directly connected door (Code 0x0e and 0x8e)
000205 to turn lights on/off? (Code 0x09)

closed 2024-08-26 by Mat931