Mat931/esp32-doorbell-bus-interface #4: Can't replicate command

dgomes, 2024-02-11:
```
[17:46:36][D][button:010]: 'Ring Indoor Doorbell' Pressed.
[17:46:36][D][remote.abbwelcome:054]: Transmitting: 55.FF.02.0A.01.01.01.01.D5.01.02.99 (12) [0101 > 0101] Type: 0A, Data: 01.02
[17:46:36][D][remote.abbwelcome:054]: Transmitting: 55.FF.02.0B.01.01.01.01.D3.81.02.32 (12) [0101 > 0101] Type: 0B, Data: 81.02
[17:46:36][I][remote.abbwelcome:110]: Received: 55.FF.02.0A.01.01.01.01.D5.01.02.99 (12) [0101 > 0101] Type: 0A, Data: 01.02
[17:46:36][I][remote.abbwelcome:110]: Received: 55.FF.02.0B.01.01.01.01.D3.81.02.32 (12) [0101 > 0101] Type: 0B, Data: 81.02
[17:46:36][I][remote.abbwelcome:110]: Received: 55.FF.42.1A.01.01.01.01.D5.01.02.80.00.FC (14) [01D501 > 010101] Type: 1A, Data: 80.00
```

actual command working (key pressed):
```
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.42.0A.00.01.01.00.01.01.46.01.02.30 (14) [000101 > 000101] Type: 0A, Data: 01.02
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.42.0B.00.01.01.00.01.01.46.81.02.9B (14) [000101 > 000101] Type: 0B, Data: 81.02
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.42.8B.00.01.01.00.01.01.46.01.02.5A (14) [000101 > 000101] Type: 8B, Data: 01.02
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.42.8A.00.01.01.00.01.01.46.8B.02.F3 (14) [000101 > 000101] Type: 8A, Data: 8B.02
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.45.01.00.01.01.00.01.01.57.09.02.01.1E.01.2A (17) [000101 > 000101] Type: 01, Data: 09.02.01.1E.01 (5)
[17:47:16][I][remote.abbwelcome:110]: Received: 55.FF.44.81.00.01.01.00.01.01.57.09.01.1E.01.15 (16) [000101 > 000101] Type: 81, Data: 09.01.1E.01
```

We can see that there is 1 byte missing 

esphome yaml:
```
  - platform: template
    name: "Ring Indoor Doorbell"
    on_press:
      - remote_transmitter.transmit_abbwelcome:
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0A # make call 
          data: [0x01, 0x02]
      - remote_transmitter.transmit_abbwelcome:
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0B # make call 
          data: [0x81, 0x02]
```

mallorca2288, 2024-02-11:
I think you are missing the three_byte_address option.

Try this:
```
  - platform: template
    name: "Ring Indoor Doorbell"
    on_press:
      - remote_transmitter.transmit_abbwelcome:
          three_byte_address: true
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0A # make call 
          data: [0x01, 0x02]
      - remote_transmitter.transmit_abbwelcome:
          three_byte_address: true
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0B # make call 
          data: [0x81, 0x02]
```

dgomes, 2024-02-11:
still didn't work:

```
[20:59:24][D][remote.abbwelcome:054]: Transmitting: 55.FF.42.0A.00.01.01.00.01.01.1A.01.02.6B (14) [000101 > 000101] Type: 0A, Data: 01.02
[20:59:24][D][remote.abbwelcome:054]: Transmitting: 55.FF.42.0B.00.01.01.00.01.01.81.81.02.E8 (14) [000101 > 000101] Type: 0B, Data: 81.02
```

mallorca2288, 2024-02-11:
Maybe you also need to transmit code 0x01 (Call)?

Try:
```
  - platform: template
    name: "Ring Indoor Doorbell"
    on_press:
      - remote_transmitter.transmit_abbwelcome:
          three_byte_address: true
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0A # make call 
          data: [0x01, 0x02]
      - remote_transmitter.transmit_abbwelcome:
          three_byte_address: true
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x0B # make call 
          data: [0x81, 0x02]
      - remote_transmitter.transmit_abbwelcome:
          three_byte_address: true
          source_address: 0x000101 # your indoor station address
          destination_address: 0x000101 # your indoor station address
          message_type: 0x01 # make call 
          data: [0x09,0x02,0x01,0x1E,0x01]
```

dgomes, 2024-02-11:
That did the trick! :) 

Thank you

dgomes, 2024-02-11:
BTW... 

after it "works" I do capture several:

```
[21:50:10][I][remote.abbwelcome:110]: Received: 55.FF.42.22.00.FF.FF.00.03.01.99.00.10.44 (14) [000301 > 00FFFF] Type: 22, Data: 00.10
```

closed 2024-02-11 by dgomes