Mat931/esp32-doorbell-bus-interface #5: [Question] PCB Housing

vr-hunter, 2024-02-11:
Hey @Mat931, thanks again for this awesome project! Mine is working nicely so far!

Since my PCB is currently dangling from the bus lines, I was wondering if you chose the PCB with an existing housing in mind? 
If not, would it be possible to share the 3D model as a step file?

Mat931, 2024-02-11:
Hi @vr-hunter!
I designed the board so it can fit in a 60mm diameter circle (flush mounted wall box) with not much room to spare. If you need a 3D model of the board you can import the PCB file into EasyEDA and export an OBJ file from there.

vr-hunter, 2024-02-12:
Gotcha, I'll try and export it from there. If I can find some time for it, I'll probably design a small wall-mountable enclosure.

closed 2024-02-12 by vr-hunter