Mat931/esp32-doorbell-bus-interface #6: [Bug] Addresses transmitted trimmed

rodricels, 2024-03-21:
First of all, thank you very much for your work releasing the design of the boards and investigating the ABB protocol, it is a magnificent job.

I am trying to run the project in a 200+ apartment community and I may have encountered a strange case with the addresses (or something I am doing terribly wrong). The addresses logged are padded with zeros but the transmission seems to remove them, the commands sent do nothing to the doors.

```
unlock_action:
  - then:
      - remote_transmitter.transmit_abbwelcome:
          source_address: 0x000110 # indoor station address, tried with 0x0110 and sends the same bytes
          destination_address: 0x000401 # building door address, tried with 0x0410 and sends the same bytes
          message_type: 0x0e # unlock door
          data: [0x00]  # door opener secret code, see receiver dump
```

```
Press the unlock button on the indoor station:
[00:37:24][D][text_sensor:064]: 'Doorbell Intercom': Sending state '55.FF.41.0E.00.04.01.00.01.10.BC.00.66 (13) [000110 > 000401] Type: 0E, Data: 00'
[00:37:24][D][text_sensor:064]: 'Doorbell Intercom': Sending state '55.FF.41.8E.00.01.10.00.04.01.BC.00.22 (13) [000401 > 000110] Type: 8E, Data: 00'

Send the unlock command
[00:37:33][D][remote.abbwelcome:054]: Transmitting: 55.FF.01.0E.04.01.01.10.2A.00.96 (11) [0110 > 0401] Type: 0E, Data: 00
```

The number of bytes are different, is it possible that the first two zeros are being removed from the addresses?
Am I sending the unlock command wrong?

Mat931, 2024-03-21:
I think you need to set `three_byte_address: true`. Here's more info: https://github.com/Mat931/esp32-doorbell-bus-interface/blob/main/how_to_customize.md#basics

rodricels, 2024-03-21:
Totally right 😓

closed 2024-03-21 by rodricels