Mat931/esp32-doorbell-bus-interface #8: Very cool project

oskarn97, 2024-10-02:
Hey Mat! 

Very cool project - I have created a similar device for another manufacturer and now switched flats - the new system is Busch Jäger Welcome. I was wondering if you still have a spare assembled version at hand. Would love to support your work.

Oskar

oskarn97, 2024-10-03:
Thanks for the reply Mat - no worries, will order it from JLC then. Well done, again!

closed 2024-10-03 by oskarn97

Pixelramp, 2025-01-14:
> Thanks for the reply Mat - no worries, will order it from JLC then. Well done, again!

Sorry for opening this again, did you order a PCB and got one left for me?

Mat931, 2025-01-14:
@oskarn97 Still no. But maybe you can organize a group order with someone else like it was done recently.

oskarn97, 2025-01-14:
I have one left as I had to order two - you can write me on Reddit u/oskarn97

oskarn97, 2025-01-20:
@Pixelramp do you still want one? You didn't reply to my message in reddit.

Pixelramp, 2025-01-21:
@oskarn97 Hey, sorry for the late reply! I’ve got some coming from JLCPCB. A neighbor also wants the PCB.