Mat931/esp32-doorbell-bus-interface #9: Wiring Diagram

maximusVII, 2024-11-05:
@Mat931: Thanks a lot for sharing your excellent code and hardware design! Would you mind sharing the wiring diagram? Would it be possible to connect it directly to an indoor station like the Busch-Jaeger 83210?

Mat931, 2024-11-05:
Hi @maximusVII 
Not sure which wiring diagram you are looking for. The schematic of the PCB is available [here](https://github.com/Mat931/esp32-doorbell-bus-interface/blob/main/design/Bus_Interface_for_ABB-Welcome_v2.pdf). There is currently no diagram on how to connect it to your Busch-Jaeger system. If your system uses only 2 wires (usually labelled `a1` and `b1`) then it's probably compatible with this project. Based on a quick search I think 83210 is compatible. You just connect the PCB to these 2 wires in parallel with your indoor station with BUS+ going to the positive and BUS- going to the negative wire.

maximusVII, 2024-11-05:
@Mat931: Thanks for your quick response! This is exactly the information I needed. It’s my first time integrating a bus system, and I wasn’t sure if I just needed to connect the ESP-doorbell-bus to a1 and b1 on the indoor station, whether it also requires a separate power source, or if it needs to be connected to another welcome device.

I’ve just ordered 5 devices—if anyone else would like one, let me know!

[image]

closed 2024-11-05 by maximusVII