Mat931/esp32-doorbell-bus-interface #10: Ready to buy hardware?

RoMaTiX99, 2024-11-14:
Hi there! 

I wonder whether anyone has a working pcb laying around at home and is willing to sell it? 
I'm not familiar with pcb creation - however I managed to process it on JLCPCB (thanks for the good description!). 

For me the price point of roughly 70$ is quite high - especially since I don't need 5 pieces. 

So if anybody has some left-overs and is willing to sell them (shipping to Germany) I'd be really grateful! 

Best regards!

bladecell, 2024-11-19:
Hey, I have just received 5 of them so I could spare one or two.. send me an email if you are interested. 
estate-woozy-broom@duck.com

RoMaTiX99, 2024-11-25:
Hi @bladecell ,

thank you - we managed to organize a group buy in the other open issue. So I don’t have any need currently. 

However I will leave this issue open so others can see your offer. 

Best!

BreBar07, 2025-05-17:
Hey all,

Just received my 5 complete boards, I only need one.

Anyone who needs one yell out happy to sell at cost + shipping, I’m based in Australia.

Thanks!

mikkelke, 2025-05-22:
Hi! 

I have four complete boards that i do not need. 
If anyone in Europe need i will sell for the cost + shipping. I'm from Denmark.

Poldo2005, 2025-05-26:
Hi @mikkelke , I'm interested in buying a complete board (or two). Could you send me the price, including shipping costs to Italy, to the address indicated in the image? 

[image]

zeratul021, 2025-06-01:
I would also join the fun if anyone is ordering in bulk to EU (I am in Czechia). Hit me up with pricing details, much appreciated:
range.view6078@eagereverest.com

Chillidaddy, 2025-06-02:
> Hi!
> 
> I have four complete boards that i do not need. If anyone in Europe need i will sell for the cost + shipping. I'm from Denmark.

Hello,
from Germany. If you have a board left I’m interested. 

popeline-sprueche-0u@icloud.com
(throwaway iCloud Email address.)

klausellus-wallace, 2025-06-03:
I'd like to join the ride too. HMU if you got spare boards flying around, for shipping in EU.[image]

harl-e, 2025-08-27:
Hello, I am looking for a board at an affordable price with shipping to Germany.
Alternatively, I am also interested in people who want to share the cost of a new order.

JTorkk, 2025-08-28:
> Hello, I am looking for a board at an affordable price with shipping to Germany. Alternatively, I am also interested in people who want to share the cost of a new order.

I have few left and can send to Germany. Send me a email at: [juho@torkke.li](mailto:juho@torkke.li)

coldhand7, 2025-11-14:
Hello @rafuz, sorry. Just yesterday I've got a board from a nice swiss guy.

sagitt, 2025-11-14:
hi, anyone have a board to sell to me? thanks
to italy

chochb, 2025-12-03:
I found one on eBay!

Hopefully it works too :)

Does anyone have a 3D printer template for a suitable case?

sagitt, 2025-12-24:
> I found one on eBay!
> 
> Hopefully it works too :)
> 
> Does anyone have a 3D printer template for a suitable case?

https://github.com/Mat931/esp32-doorbell-bus-interface/issues/26

Here

zwinsch64, 2026-01-15:
first: so many thanks to Mat931 for sharing his great project! Awesome!
(I would love to have also Audio and Video... but this seems to be quiet a task, following what can be found in the web)

at adamjez et all:
I did order 5 Boards from JLCPCB on Dec. 30th and they did arrive today. 
I am willing to give away 3 to 4 of them. (lets say: 20€ plus shipping costs - from Frankfurt a/M, Germany)
The boards are fully assembled **but: WITHOUT the ESP** and without the Bus-Header!
I also have spare ESP Solo1 and Headers - so in case you really need it: I could additionally solder them (by hand) to the board.

SetupCoding, 2026-02-05:
I ordered 5 boards and was able to make everything work with the first one on my BUSCH-JAEGER 83205 AP-6 intercom.
Flashing was a learning curve though.

Thus I have 4 boards with female jumpers (P2) and an ESP32 module each for sale.

Shipping inside EU (from Germany) as a tracked Parcel via DHL 6,19€.

If you are interested I am willing to sell a board for 30€ + shipping.

Feel free to reach out to me: SetupCoding@dev.nondon.store (temporary address for the next two months).

[image]

zwinsch64, 2026-04-18:
> @asiersan https://github.com/asiersan
> With ESP and Headers soldered, i cant see your e-mail here.
> Done, and mail sent

Hallo @asiersan,

I did not receive any, neither email with postal address, nor ppal payment bey now.
Sure you did use my address as shown above (karsten dot f dot w at gmx dot de)?

javicalle, 2026-04-19:
Anyone interested in a group buy? I’m located in Spain and happy to ship within the EU.

I'm interested in the device and have started looking into how to order it. For now, it seems the ESP32s are unavailable. From the photo, it looks like I could get the other one they do have available (ESP32-SOLO-1(4MB)
C2934562), but comparing the PCB footprint diagram is confusing. Can anyone confirm if it's possible to swap them?

SetupCoding, 2026-04-26:
Hey @threadstone83 I still have plenty devices (pre flashed for easy provisioning) with ESP32 modules on them and am shipping in Europe. If you want to buy one (30€ + tracked DHL shipping), feel free to contact me with your shipping address: anton.setup.schmidt+abb@gmail.com

fquinto, 2026-08-25:
Hi everyone,

I’ve designed a new, updated version of the electronic board for this project. The design is ready, but **the board has not been manufactured or tested yet**, so it should be considered an **untested prototype**.

I’m looking for people who might be interested in joining a **group buy / small production run** to have the boards manufactured and then test them together.

Project: https://github.com/fquinto/abb-welcome-gateway
Contact / discussion: https://t.me/abb_welcome_gateway

If you’re interested in joining the group buy or helping with testing, please let me know!

TUTAMKHAMON, 2026-10-06:
Hey all. I have 4 extras to sell if anybody is interested.
Shipping from Copenhagen to anywhere in Europe.

SetupCoding, 2026-11-05:
Hello everyone,
I still have a couple of pre-flashed boards with esp32 installed.
I am shipping inside EU (from Germany) as a tracked + insured Parcel via DHL.

If you are interested I am willing to sell a board for 30€ + shipping.

Feel free to reach out to me: [anton.setup.schmidt+abb@gmail.com](mailto:anton.setup.schmidt+abb@gmail.com)

Additionally: I always wanted to have a write up of my journey with automating the doorbell, so here goes, if anyone is interested: https://setup-coding.de/blog/busch-jaeger-home-assistant

And full build guide: https://setup-coding.de/blog/busch-jaeger-home-assistant-how-to

[image]