Mat931/esp32-doorbell-bus-interface #12: Switched BUS+ and BUS-

bladecell, 2024-12-16:
Hey, I am having slight issue with my doorbell, that has polarity of BUS A and B switched meaning that BUS+ is B and BUS- is A. That results in the device not powering on when connected. 

When switching cables from my doorbell, connecting BUS A to the BUS- (B) on the board and BUS B to the BUS+ (A) on board the device boots up but when connected simultaneously with the "phone" part of the doorbell it does not receive any messages in log or in HA. I can send messages through HA and see them in logs being transieved but not having my door opened or anything else.

I have tried switching TX and RX pin in the yaml with the switched connection but without any luck, still got none messages received.

So I am wondering if it should work with the switched pins in yaml and the issue is with the board, that I have soldered myself or the issue it somewhere else.

This is my doorbell for reference:

[image]

Mat931, 2024-12-16:
Hi @bladecell 
If the power LED on my board lights up, it's connected correctly. There is no other way to connect it. If the LED doesn't light up, swap the wires. You should not swap the TX and RX pin numbers in the YAML!
With everything connected, try pressing the door opener on your indoor station. You should receive 2 messages. Check the log in ESPHome to see them. From there you can start [customizing the yaml for your system](https://github.com/Mat931/esp32-doorbell-bus-interface/blob/main/how_to_customize.md).

bladecell, 2024-12-16:
I can make the LED light up but I have to connect the b connector on my doorbell to the BUS+ (A) connector on the board and the a connector on the doorbell to the BUS- (B) on the board, otherwise it will not power on.

Then with the "ESPHome Example Configuration" verbatim I am not receiving any messages in the logs or in the HA entity.

Mat931, 2024-12-16:
It's normal for some systems to have A and B swapped, that can't be the reason why you're not receiving anything.
If you soldered the board yourself, you could check the connections and component values, especially on the receiver circuit.

bladecell, 2024-12-18:
Had to resolder a few component but the receiving works now, thanks for the tip.

closed 2024-12-19 by Mat931