Mat931/esp32-doorbell-bus-interface #13: Board not booting / connecting to wifi

RoMaTiX99, 2024-12-24:
Hello everyone, 

I received a PCB that had a broken capacitor (C23) - at least it was lose. During the flashing process I wasn't able to get a connection; however when I (only) connected 18V to the +/- BUS Pins there was no LED flashing (not even the Power LED). So I thought the board would be cracked due to the broken capacitor. 

Today I took the time for some further inspections. Here's what I did and what I found out: 

- I resoldered the broken capacitor 
- I connected 18V DC to the BUS Pins - still no LEDs showing up, not even power LED 
- I took external but very stable 5V (and gnd ofc) directly to the ESP Pins and started the flashing process 
- Flashing inside HomeAssistant with ESPHome failed; it was not even able to establish a connection 
- I downloaded the Make File and flashed it via web.esphome.io; surprisingly this worked! 
- After I successfully flashed the board I was looking into the logs since I expected the board to connect with my wifi or give me a fallback hotspot 
-  neither was the case - and I can't find anything relevant in the logs (which I can access when I connect the 5V to ESP Pin with the adapter). 

The log file is attached [logs (5).txt](https://github.com/user-attachments/files/18240945/logs.5.txt); including 3 (failed) boot iterations. During one boot the log shows the following multiple times, which indicated an hardware error to me: 
`[16:44:11]rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:44:11]flash read err, 1000
[16:44:11]ets_main.c 371 
[16:44:11]ets Jun  8 2016 00:22:57`

For me it looks like the boot process is stuck by loading wifi configuration / modulare: 
```
[C][wifi:062]:   Loca ets Jun  8 2016 00:22:57
[16:47:18]
[16:47:18]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:47:18]configsip: 0, SPIWP:0xee
[16:47:18]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:47:18]mode:DIO, clock div:2
[16:47:18]load:0x3fff0030,len:6644
[16:47:18]load:0x40078000,len:15056
[16:47:18]ho 0 tail 12 room 4
[16:47:18]load:0x40080400,len:3628
[16:47:18]entry 0x40080668
[16:47:18]I (28) boot: ESP-IDF 4.4.8 2nd stage bootloader
[16:47:18]I (28) boot: compile time 16:38:51
[16:47:18]W (28) boot: Unicore bootloader
[16:47:18]I (31) boot: chip revision: v1.1
[16:47:18]I (35) boot.esp32: SPI Speed      : 40MHz
[16:47:18]I (40) boot.esp32: SPI Mode       : DIO
[16:47:18]I (44) boot.esp32: SPI Flash Size : 4MB
[16:47:18]I (49) boot: Enabling RNG early entropy source...
[16:47:18]I (54) boot: Partition Table:
[16:47:18]I (58) boot: ## Label            Usage          Type ST Offset   Length
[16:47:19]I (65) boot:  0 otadata          OTA data         01 00 00009000 00002000
[16:47:19]I (72) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[16:47:19]I (80) boot:  2 app0             OTA app          00 10 00010000 001c0000
[16:47:19]I (87) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[16:47:19]I (95) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[16:47:19]I (102) boot: End of partition table
[16:47:19]I (107) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=21cd4h (138452) map
[16:47:19]I (165) esp_image: segment 1: paddr=00031cfc vaddr=3ffb0000 size=02b00h ( 11008) load
[16:47:19]I (170) esp_image: segment 2: paddr=00034804 vaddr=40080000 size=0b814h ( 47124) load
[16:47:19]I (190) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8d930h (579888) map
[16:47:19]I (400) esp_image: segment 4: paddr=000cd958 vaddr=4008b814 size=089f4h ( 35316) load
[16:47:19]I (425) boot: Loaded app from partition at offset 0x10000
[16:47:19]I (425) boot: Disabling RNG early entropy source...
[16:47:19]I (436) cpu_start: Unicore app
[16:47:19]I (437) cpu_start: Pro cpu up.
[16:47:19]I (437) cpu_start: Single core mode
[16:47:19]I (448) cpu_start: Pro cpu start user code
[16:47:19]I (449) cpu_start: cpu freq: 160000000
[16:47:19]I (449) cpu_start: Application information:
[16:47:19]I (453) cpu_start: Project name:     abb-welcome
[16:47:19]I (458) cpu_start: App version:      2024.11.3
[16:47:19]I (464) cpu_start: Compile time:     Dec 24 2024 16:38:19
[16:47:19]I (470) cpu_start: ELF file SHA256:  35fab73e87e09151...
[16:47:19]I (476) cpu_start: ESP-IDF:          4.4.8
[16:47:19]I (480) cpu_start: Min chip rev:     v0.0
[16:47:19]I (485) cpu_start: Max chip rev:     v3.99 
[16:47:19]I (490) cpu_start: Chip rev:         v1.1
[16:47:19]I (495) heap_init: Initializing. RAM available for dynamic allocation:
[16:47:19]I (502) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[16:47:19]I (508) heap_init: At 3FFB6BB8 len 00029448 (165 KiB): DRAM
[16:47:19]I (514) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[16:47:19]I (521) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
[16:47:19]I (527) heap_init: At 40094208 len 0000BDF8 (47 KiB): IRAM
[16:47:19]I (533) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
[16:47:19]I (541) spi_flash: detected chip: generic
[16:47:19]I (544) spi_flash: flash io: dio
[16:47:19]I (549) cpu_start: Starting scheduler on PRO CPU.
[16:47:26][I][logger:171]: Log initialized
[C][status_led:014]: Setting up Status LED...
[D][esp-idf:000]: I (754) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[C][safe_mode:079]: There have been 3 suspected unsuccessful boot attempts
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][lock:054]: 'Front Door': Sending state LOCKED
[C][remote_receiver.esp32:013]: Setting up Remote Receiver...
[D][esp-idf:000]: I (788) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

[C][wifi:048]: Setting up WiFi...
[D][esp-idf:000]: I (802) wifi:
[D][esp-idf:000]: wifi driver task: 3ffc6900, prio:23, stack:6656, core=0
[D][esp-idf:000]: 

[D][esp-idf:000][wifi]: I (819) system_api: Base MAC address is not set

[D][esp-idf:000][wifi]: I (828) system_api: read default base MAC address from EFUSE

[D][esp-idf:000][wifi]: I (841) wifi:
[D][esp-idf:000][wifi]: wifi firmware version: ff661c3
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (859) wifi:
[D][esp-idf:000][wifi]: wifi certification version: v7.0
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (869) wifi:
[D][esp-idf:000][wifi]: config NVS flash: enabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (890) wifi:
[D][esp-idf:000][wifi]: config nano formating: disabled
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (900) wifi:
[D][esp-idf:000][wifi]: Init data frame dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (921) wifi:
[D][esp-idf:000][wifi]: Init static rx mgmt buffer num: 5
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (941) wifi:
[D][esp-idf:000][wifi]: Init management short buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (951) wifi:
[D][esp-idf:000][wifi]: Init dynamic tx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (972) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer size: 1600
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (992) wifi:
[D][esp-idf:000][wifi]: Init static rx buffer num: 10
[D][esp-idf:000][wifi]: 

[D][esp-idf:000][wifi]: I (1003) wifi:
[D][esp-idf:000][wifi]: Init dynamic rx buffer num: 32
[D][esp-idf:000][wifi]: 

[D][esp-idf:000]: I (1024) wifi_init: rx ba win: 6

[D][esp-idf:000]: I (1024) wifi_init: tcpip mbox: 32

[D][esp-idf:000]: I (1043) wifi_init: udp mbox: 6

[D][esp-idf:000]: I (1044) wifi_init: tcp mbox: 6

[D][esp-idf:000]: I (1054) wifi_init: tcp tx win: 5760

[D][esp-idf:000]: I (1054) wifi_init: tcp rx win: 5760

[D][esp-idf:000]: I (1064) wifi_init: tcp mss: 1440

[D][esp-idf:000]: I (1064) wifi_init: WiFi IRAM OP enabled

[D][esp-idf:000]: I (1074) wifi_init: WiFi RX IRAM OP enabled

[C][wifi:061]: Starting WiFi...
[C][wifi:062]:   
````

Has anybody had similar problems? 
Or does anyone have any idea what the problem might by / which part may be broken? 

For me it looks like the broken capacitor is no longer the problem since I was able to measure 18V there. Since I was able to flash the board I guess it might be a problem elsewhere.... 

Any help is appreciated! 

Thanks in advance. 

Best!

Mat931, 2024-12-24:
Hi @RoMaTiX99 

Did you really connect 5V directly to the ESP32? This could have damaged it since its maximum voltage is 3.6V. Ideally you should use 3.3V to power the ESP32 directly.

The log looks like the ESP32 crashes when it attempts to connect to WiFi. Maybe your power supply can't deliver the current required for WiFi. Does the crash happen when powering the ESP32 directly or are you using the BUS pins and onboard step-down converter?

Maybe there is still a problem with the step-down converter like a broken connection to one of its capacitors. That could explain an unstable voltage that could cause the crash.

RoMaTiX99, 2024-12-25:
Hi @Mat931,

your response times are amazing! Thanks a lot and happy Christmas.

Yes I connected 5V since I’ve never bricked and esp32 by this. Probably this was dumb… 

I supply voltage by the esp32 pins when it powers up. 
As written, the board won’t power up when I connect 18V DC to the bus pins. 

Any idea why this is? So why is there no led when I connect the bus pins. 

Best,
Timon

Mat931, 2024-12-25:
For this next step you need a multimeter. Put your multimeter into voltage mode and connect the negative lead to BUS- and the positive to each of the 5 spots in the picture. Power the board over BUS+/BUS- while you measure the voltages. With everything working correctly, 1-4 should be close to your input voltage and 5 should be close to 3.3V. Please tell me the 5 voltages you measured. 

Then put your multimeter into resistance mode and measure the resistance between each point and BUS-, this time without powering the board. This can help us find short circuits.

[image]

RoMaTiX99, 2024-12-26:
Hi @Mat931 ,

here are the results: 
Voltage is ok - it fits to the values it should have. 

Resistancy:

1: 0,4 kOhm 
2: 0,4 kOhm 
3: 5,5 MOhm 
4: 5,5 MOhm 
5: 30 kOhm, but decreasing rapidly, going down to 4 kOhm, probably bc it’s charging a capacitor? 

is this anything unusual? 

Thanks in advance!

Mat931, 2024-12-26:
In that case I think your ESP32 might be damaged or there's still a broken connection to a capacitor (maybe C23).
If all voltages are ok, the power LED should light up. Is that correct?

RoMaTiX99, 2024-12-26:
Sadly the PowerLED doesn’t light up.

Mat931, 2024-12-26:
Did it ever light up? Maybe it's soldered in reverse or there's a broken trace somewhere.
I'm running out of ideas what could be wrong with your board. Do you have a second board to try?

RoMaTiX99, 2024-12-26:
No, it never lighted up. 
I think it’s unlikely that it’s soldered in reverse; it was part of a group buy and it seems like only my board is broken. 

Sadly I don’t have another board to try. But if there’s no other way I will order another 5 and give it a try…

Mat931, 2024-12-26:
If your board arrived with a broken capacitor and other problems you can try to get a refund from JLC.
One last thing you can try is to power the board over the BUS pins and measure the voltage on the programming header between 3V3 and GND. If you get 3.3V on position 5 but not on the programming header, you have a broken trace. If there's 3.3V even on the programming header then maybe the LED is broken.

RoMaTiX99, 2025-01-03:
Hi. I have to check, but LED can’t be broken since it lights up when connected via the programming header. 

However one last question: what should be the value of the capacitor? I measured 79yF. Is this correct?

Mat931, 2025-01-03:
If you mean C23, that is just outside the range of acceptable values (100µF ±20% -> 80-120µF)
But if that was the only problem, the ESP32 would still work.

RoMaTiX99, 2025-01-06:
Hi again. 
I just re-checked everything and I have to correct the measurements of the resistancy:

1: 550 kOhm
2: 540 kOhm
3: Open
4: Open
5: 30kOhm>4kOhm decreasing 

and there is more: 
I noticed, that on the other capacitor (not the broken one) there was no voltage on the backside pins. So it seemed to have no connection. I wired it to the 3.3V pin of the esp32 and now the WiFi Led did light up for the first time. Any idea? 

Next step for me is to check, whether I can get new logs with this setup and what they say (since I can’t see the device in my WiFi nor the fallback hotspot).

RoMaTiX99, 2025-01-11:
Hi @Mat931,

thanks again for all you awesome support. Finally I can give some good news:
I got my board working! 

I connected the capacitor to the esp32 3.3V pin so it has stable power connection. Now everything seems to work fine. Will start configuring now. I will close this issue now. Thanks a lot!!!

closed 2025-01-11 by RoMaTiX99