Mat931/esp32-doorbell-bus-interface #14: Failed config with ESPHome 2025.2.0

d-l-h, 2025-02-24:
With `ESPHome 2025.2.0` configuration is failing

    `Failed config
    
    This feature is only available with frameworks ['arduino'].
    memory_blocks: 5
    
    This feature is only available with frameworks ['arduino'].
    clock_divider: 160
    on_abbwelcome: 
      then:  `

Mat931, 2025-02-24:
I think it should be okay to remove these lines (`memory_blocks` and `clock_divider`). Can you try if it still works with the lines removed? If that doesn't work you can either change the framework to `arduino` or try an ESP-IDF version below 5.0.0.

d-l-h, 2025-02-25:
>remove these lines (`memory_blocks` and `clock_divider`)

This worked fine! Thank you. I created a PR.

Mat931, 2025-02-25:
Thanks for testing!

closed 2025-02-25 by Mat931