Mat931/esp32-doorbell-bus-interface #17: Question: any list of verified / supported devices

zeratul021, 2025-06-01:
Project looks great! Went through the docs and was wondering, if there are success stories of installation where it works, including pictures of final wiring and setup. I want to order the board and try it out with my M22002-W-02 model, would love to make it work and contribute to such wiki page.

Mat931, 2025-06-02:
Thanks! I created a [wiki page](https://github.com/Mat931/esp32-doorbell-bus-interface/wiki/Supported-devices). If anyone else has a working device I will add it to the list.

zeratul021, 2025-06-03:
@fischmal thanks! Mine looks like this

[image]

fischmal, 2025-06-03:
> [@imuncas](https://github.com/imuncas) Personally I have [this red one](https://www.amazon.de/AZDelivery-Adapter-FT232RL-Serial-gratis/dp/B01N9RZK6I), but it doesn't provide enough power for the ESP32 so you need to power the board externally.

I have also used this serial adapter. Worked well with jumper wires ([amazon link](https://www.amazon.de/dp/B01EV70C78)) and the adapter delivered enough power in my case (set the jumper to 3.3).

[image]

zeratul021, 2025-06-28:
@Mat931 I managed to (with many difficulties due to serial adapter failing to hold 3.3 V, had to use external 12V adapter) flash esp-home and wire the component. So far I haven't been able to catch code for opening doors, but stop ringing button works.
Also, device uses 3 byte addresses, you can update the wiki.

fquinto, 2025-09-30:
@zeratul021 Can you check this?

### RFID Key Messages Found

**Type 0x35 (Short RFID format):**

    Destination: 0x001301 (likely an access controller)

    Data: 02.00 (appears to be a simple code/identifier)

    Example: 55.FF.42.35.00.13.01.00.02.02.7D.02.00.99

**Type 0x60 (Long RFID format with full UID):**

    Destination: 0x000E01 (another access controller)

    Data: 02.01.13.F1.71.B6.00.00.01.01 (10 bytes)

    RFID UID appears to be: 13 F1 71 B6 (the actual chip ID)

    Example: 55.FF.4A.60.00.0E.01.00.02.02.03.02.01.13.F1.71.B6.00.00.01.01.FE

Where the RFID Key Data Is:

In the Type 0x60 message, the RFID key's unique identifier is clearly visible in the data section:

-     Bytes 4-7: 13 F1 71 B6 - This is the actual RFID chip UID
-     The surrounding bytes appear to be facility codes, system identifiers, or other metadata

The Type 0x35 message seems to be a simplified version that might use a shorter, system-specific code rather than the full RFID UID.

So when someone uses an RFID key fob at your front door, the system broadcasts both a short code (0x35) to one controller and the full RFID UID (0x60) to another controller, likely for different purposes like logging vs. immediate access control.

Your RFID key's UID is: **13F171B6**
or maybe **334590390** in RFID print card

Mat931, 2026-02-08:
@SetupCoding Thanks! I've added it to the page.

ChrisDietrich, 2026-09-10:
Great project, thanks a lot! I can confirm Busch-Jaeger 83210 works fine as well.

This line is for the https://github.com/Mat931/esp32-doorbell-bus-interface/wiki/Supported-devices table, in case it's useful for others.

| `Busch-Jaeger 83210` | Audio indoor station | a₁, b₁ | 2 | `false` |